PlotJuggler/plotjuggler-ros-plugins #2: Subscribing to MultiArray messages

sophiaas, 2020-12-16:
In the earlier version of plotjuggler, I was able to subscribe to topics that published Float32MultiArrays. On the sidebar, the data would show up with each index of the array, e.g. data.0, data.1, etc, and I was able to plot all indices as line plots simultaneously. This was awesome! I was using it all the time for one of my projects.

After installing the new version, the same approach no longer works. It appears that plotjuggler is not reading the full contents of the Float32MultiArray message. In the sidebar, it shows the "layout" attribute, which contains the message index and "data_offset" (a scalar), but the "data" attribute which contains my array no longer shows up. I have verified that "data" is indeed an attribute of the message, but it is apparently invisible to plot juggler.

Would be awesome to have this capability back!

facontidavide, 2020-12-16:
I think the problem in not that. Both the old and new version might reject arrays that are very long. Double check the length of your array.

[image]

sophiaas, 2020-12-17:
Wow, thanks so much for catching that - solved it.

closed 2020-12-18 by facontidavide