PlotJuggler/plotjuggler-ros-plugins #5: Quaternion plugin does not wrap euler angles

romainreignier, 2021-01-15:
While plotting an imu message, I have noticed that the `yaw_deg` field is not wrapped between -180° and +180° but keep increasing.

On the v2 branch, the issue is not there.

This commit introduced the issue: https://github.com/facontidavide/PlotJuggler/commit/0dd18a143bb70606acb85a4d7e557d4536e429ff

But I don't really understand the purpose of that commit so I did not provide a patch.

facontidavide, 2021-01-15:
no, it doesn't

facontidavide, 2021-01-15:
The reason is that I like it this way.

If your robot does three rotations, I want to see that.

romainreignier, 2021-01-15:
Humm ok. That's surprising but now the commit makes sense.

facontidavide, 2021-01-15:
Ideally, we make this configurable. I will keep this open as a "nice to have"

ljburtz, 2022-07-19:
[image]
@facontidavide hello, hope this is the right place
I have a noisy IMU which seems to trigger the WRAP_ANGLE / WRAP_THRESHOLD logic in the commit repeatedly for the yaw angle, even while the rover is not moving (see screenshot).
What would you recommend to be the best way to have the yaw angle stay between 0 and 2Pi?