PlotJuggler/plotjuggler-ros-plugins #8: Plotjuggler crashes after subscribing to ROS topic (`Floating point truncated 1`)

nspo, 2021-01-26:
I'm using Plotjuggler 3.0.7 (Snap) on Ubuntu 18.04. Each time I subscribe to a specific ROS topic, Plotjuggler crashes immediately. It's possible other "bad" topics exist that make it crash but I identified at least one that's enough.

How to reproduce:
- Start `roscore`
- Replay bag file with a "bad" topic: `rosbag play -l filename.bag`
- Start ROS subscriber and select the "bad" topic
- Plotjuggler crashes with the attached error messages

[plotjuggler-crash-after-subscribing-ROS-topic.txt](https://github.com/PlotJuggler/plotjuggler-ros-plugins/files/5873251/plotjuggler-crash-after-subscribing-ROS-topic.txt)

A similar thing happens when trying to open the bag file directly without streaming and selecting the topic. In this case it catches the exception though:

[image]

facontidavide, 2021-01-26:
Ok. It is NOT supposed to crash (I need to fix this), but probably this is a duplicated issue:

https://github.com/facontidavide/PlotJuggler/issues/325

nspo, 2021-01-26:
In case this has the same root cause (data in bag cannot be displayed due to integer values needing too many bits), would it not make sense to optionally allow this ("Allow integer adulteration? [yes/no]")?

facontidavide, 2021-02-27:
ploblem should be fixed

closed 2021-02-27 by facontidavide

nspo, 2021-02-27:
Great, thanks! Can't verify yet as the snap is still on 3.0.7 but will do when it's updated.