PlotJuggler/plotjuggler-ros-plugins #10: Int64 type makes PlotJuggler crash

MCFurry, 2021-02-26:
To easily reproduce `rostopic pub -r10 /test std_msgs/Int64 "data: 0"` and let PlotJuggler subscribe to that topic, it will immediately crash.
If I import a bagfile with the same datatype message, it throws me `Variant::extract -> wrong type`

Doing the same tests with std_msgs/UInt64 seems to work however? Any reason why the Int64 datatype gives problems?

Timple, 2021-02-26:
Same here. For what it's worth:

```
Loading compatible plugins from directory:  "/opt/ros/melodic/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamMQTT.so" : is a DataStreamer plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libDataStreamUDP.so" : is a DataStreamer plugin
"libDataStreamWebSocket.so" : is a DataStreamer plugin
"libDataStreamZMQ.so" : is a DataStreamer plugin
Number of plugins loaded:  7 

Loading compatible plugins from directory:  "/opt/ros/melodic/lib/plotjuggler_ros"
"libDataLoadROS.so" : is a DataLoader plugin
"libDataStreamROS.so" : is a DataStreamer plugin
"libRosoutPublisher.so" : is a StatePublisher plugin
"libRosTopicPublisher.so" : is a StatePublisher plugin
Number of plugins loaded:  4 

Loading compatible plugins from directory:  "/home/tim/.local/share/PlotJuggler"
Number of plugins loaded:  0 

terminate called after throwing an instance of 'RosIntrospection::TypeException'
  what():  Variant::extract -> wrong type
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c60b7d9, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c60916a, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c5b6631, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2f4b62d8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c5b765e, in QApplication::notify(QObject*, QEvent*)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c5af83b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c5ee047, in QWidget::event(QEvent*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c6a238c, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c6a2199, in QAbstractButton::setIconSize(QSize const&)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c6a0db9, in QAbstractButton::toggled(bool)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2f4c6a0ba1, in QAbstractButton::clicked(bool)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2f4b65c618, in QMetaObject::activate(QObject*, int, int, void**)
#19   Object "/opt/ros/melodic/lib/plotjuggler/plotjuggler", at 0x55d840a05298, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const
#18   Object "/opt/ros/melodic/lib/plotjuggler/plotjuggler", at 0x55d840a0417d, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const
#17   Object "/opt/ros/melodic/lib/plotjuggler/plotjuggler", at 0x55d840965d74, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::swap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)
#16   Object "/opt/ros/melodic/lib/plotjuggler/plotjuggler", at 0x55d840965875, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::swap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)
#15   Object "/opt/ros/melodic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f2f04c798b1, in DataStreamROS::start(QStringList*)
#14   Object "/opt/ros/melodic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f2f04c785ea, in DataStreamROS::extractInitialSamples()
#13   Object "/opt/ros/melodic/lib/libroscpp.so", at 0x7f2f0d051b0a, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#12   Object "/opt/ros/melodic/lib/libroscpp.so", at 0x7f2f0d04fd98, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#11   Object "/opt/ros/melodic/lib/libroscpp.so", at 0x7f2f0d0a5421, in ros::SubscriptionQueue::call()
#10   Object "/opt/ros/melodic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f2f04c89d51, in ros::SubscriptionCallbackHelperT<boost::shared_ptr<RosIntrospection::ShapeShifter const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&)
#9    Object "/opt/ros/melodic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f2f04c80b01, in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<RosIntrospection::ShapeShifter const> const&)>, void, boost::shared_ptr<RosIntrospection::ShapeShifter const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<RosIntrospection::ShapeShifter const>)
#8    Object "/opt/ros/melodic/lib/plotjuggler_ros/libDataStreamROS.so", at 0x7f2f04c77bf7, in DataStreamROS::topicCallback(boost::shared_ptr<RosIntrospection::ShapeShifter const> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[... cut ...]
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f2f4ae84ae5, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f2f4ae7e956, in 
#1    Source "/build/glibc-S9d2JN/glibc-2.27/stdlib/abort.c", line 79, in __GI_abort [0x7f2f4a48b920]
#0    Source "/build/glibc-S9d2JN/glibc-2.27/signal/../sysdeps/unix/sysv/linux/raise.c", line 51, in __GI_raise [0x7f2f4a489fb7]
Aborted (Signal sent by tkill() 21863 1000)
Aborted (core dumped)
```

facontidavide, 2021-02-27:
fixed

closed 2021-02-27 by facontidavide

MCFurry, 2021-02-27:
Thx for the quick fix!