PlotJuggler/plotjuggler-ros-plugins #12: PlotJuggler 3.1.0 doesnt exit on Ctrl+C if subscribed to ROS topic

mikaelarguedas, 2021-04-17:
Maybe it is expected behavior, but I see different behavior in the following scenarios. I always start plotjuggler from a terminal using `rosrun plotjuggler plotjuggler`:
1. hit Ctrl+C, plotjuggler exits as expceted
1. subscribe to a ROS topic and then hit Ctrl+C => Plotjuggler stays opened
1. subscribe to a ROS topic, unsubscribe hitting the Stop button and then hit Ctrl+C => Plotjuggler stays opened
```
rosrun plotjuggler plotjuggler
```
But if I subscribe to a ROS topic

facontidavide, 2021-06-03:
I am aware of this issue and I tried to solve it... unsuccessfully so far :disappointed:

facontidavide, 2023-12-21:
Closing this for one of the following reasons:

- obsolate
- you didn't provide an easy way to reproduce it
- You should test latest version (PlotJuggler 3.8.4 + ROS plugins 2.0.0)

closed 2023-12-21 by facontidavide

mikaelarguedas, 2023-12-21:
Assuming this is closed because fixed but the fix is not yet available on the main noetic apt repo..

If this is not the case, could you please give more context for closing?

If this is the case and regardless, thanks for the continuous improvements and bug fixes on this project!