PlotJuggler/plotjuggler-ros-plugins #17: plotjuggler build from source - ros2 foxy

cedros23, 2021-06-09:
Hello,

Thanks for the excellent toolchain.
Is there a way to compile juggler from source with colcon?
When I try a luck attempts with colcon build instead of catkin_make, I have receive errors about  package.xml
 related errors caused by Juggler package configuration is designed for catkin?

thanks in advance

cedros23, 2021-06-09:
Just to close my issue, perhaps still be helpful for the other trying to build it from source, following changes required for the package.xml inside jugger msgs:

<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
    <name>plotjuggler_msgs</name>
    <version>0.1.1</version>
    <description>Special Messages for PlotJuggler</description>
    <maintainer email="davide.faconti@gmail.com">daf</maintainer>
    <license>MIT</license>

    <buildtool_depend>ament_cmake</buildtool_depend>
    <depend>message_generation</depend>
    <depend>message_runtime</depend>
    <depend>std_msgs</depend>
    <depend>builtin_interfaces</depend>
    <depend>rosidl_default_generators</depend>
    <member_of_group>rosidl_interface_packages</member_of_group>
    <export>

    </export>
</package>

closed 2021-06-09 by cedros23

facontidavide, 2021-06-09:
For everybody else:

There is already a ros2 branch for that

https://github.com/PlotJuggler/plotjuggler_msgs/tree/ros2