PlotJuggler/plotjuggler-ros-plugins #18: How to install plotjuggler with ros-plugins on ubuntu?

linnpl, 2021-06-11:
Hi, 
I am very new to Linux OS and I tried to run this command in a terminal

`sudo apt install ros-${ROS_DISTRO}-plotjuggler
`
but I got this error:
`E: unable to locate package ros--plotjuggler`
Could anyone kindly offer me an insight on what I am doing wrong? Perhaps I need to clone the repository first? My ubunbu version is 20.04

clydemcqueen, 2021-06-11:
Replace `${ROS_DISTRO}` with the name of your ROS version, e.g., `noetic` or `foxy`.

linnpl, 2021-06-12:
Thank, unfortunately I still got the same error:
`E: unable to locate package ros--plotjuggler`

linnpl, 2021-06-12:
Thanks, unfortunately I still go the error:
`E: unable to locate package ros--plotjuggler`

linnpl, 2021-06-12:
Update: I tried to compile from source code, but when calling this command
 `rosdep install --from-paths src --ignore-src --rosdistro=melodic -y`

I got this error:

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
plotjuggler_msgs: No definition of [std_msgs] for OS version [focal]
plotjuggler_ros: No definition of [tf2_msgs] for OS version [focal]
`

clydemcqueen, 2021-06-12:
My mistake, the correct install instruction is `sudo apt install ros-melodic-plotjuggler-ros` (note the trailing `-ros`).

Make sure you have run `source /opt/ros/melodic/setup.bash` to set up your environment before building or running ROS packages.

facontidavide, 2021-06-12:
@clydemcqueen thanks a lot for helping with the issue

closed 2021-06-18 by facontidavide