PlotJuggler/plotjuggler-ros-plugins #25: how to build it the plugins on windows10 ?

xiaochen-liu, 2021-07-03:
i would like to use Plotjugger  on windows to show the ros2 infomation. but the current Plotjugger 3.1.1 release don't have the ros2 plugins.  when can release the Ploljugger with ros2 support for windows10 ? or how can i build it?  thanks very much

xiaochen-liu, 2021-07-03:
more information:
i use the  colcon build --merge-install to build under windows， get the following error.

D:\ROS2\ros2-windows\include\rclcpp/message_memory_strategy.hpp(57): message: View references to alias template instantiation 'rclcpp::allocator::Deleter<std::allocator<char>,char>' being compiled (compile source file D:\ROS2\workspace\ws_plotjuggler\build\plotjuggler_ros\plugins\DataStreamROS2_autogen\mocs_compilation_Release.cpp) [D:\ROS2\workspace\ws_plotjuggler\build\plotjuggler_ros\plugins\DataStreamROS2.vcxproj]
Creating library D:/ROS2/workspace/ws_plotjuggler/build/plotjuggler_ros/plugins/Release/DataStreamROS2.lib and object D:/ROS2/workspace/ws_plotjuggler/build/plotjuggler_ros/plugins/Release/DataStreamROS2.exp
datastream_ros2.obj: error LNK2001: Unparsable external symbol "public: virtual void __cdecl rosbag2_transport::GenericSubscription::handle_loaned_message(void *, class rclcpp::MessageInfo const &)" (?handle_loaned_message@GenericSubscription@rosbag2_transport@@UEAAXPEAXAEBVMessageInfo@rclcpp@@@Z) [D:\ROS2\workspace\ws_plotjuggler\build\plotjuggler_ros\plugins\DataStreamROS2.vcxproj]
D:\ROS2\workspace\ws_plotjuggler\build\plotjuggler_ros\bin\Release\DataStreamROS2.dll: fatal error LNK1120: 1 unresolvable external commands [D:\ROS2\workspace\ws_plotjuggler\build\plotjuggler_ros\plugins\DataStreamROS2.vcxproj]
Automatic MOC and UIC for target TopicPublisherROS2
Building Custom Rule D:/ROS2/workspace/ws_plotjuggler/src/plotjuggler-ros-plugins/plugins/CMakeLists.txt
publisher_ros2.cpp
mocs_compilation_Release.cpp

windelbouwman, 2021-11-01:
I tried to compile the ros-plugins on windows. This is not yet smooth-sailing. Here are my steps:

- Download ROS2 for windows (binary zip file). I chose galactic.
- Download Qt5 for MSVC2019, this should match closely the Qt5 version bundled with ROS2.
- Download BOOST, I used version 1.75
- Install visual studio 2019

Prepare a colcon workspace:

    mkdir juggler_ws
    cd juggler_ws
    mkdir src
    cd src
    git clone https://github.com/PlotJuggler/plotjuggler-ros-plugins.git -b rolling
    git clone https://github.com/PlotJuggler/plotjuggler_msgs.git
    git clone https://github.com/facontidavide/PlotJuggler.git
    cd ..

Make some modifications:
- Add `_USE_MATH_DEFINES` to the `commonROS` target_compile_definitions. This enable `M_PI` and friends in windows.
- Remove `DataStreamROS2` completely.
- Remove the `ToolboxQuaternion` plugin, it did not compile for me.

Hit build:
    c:\program files\microsoft\ ...... \vcvarsall.bat x64
    set Qt5_DIR=C:\Qt\5.15.2\msvc2019_64
    set BOOST_ROOT=C:\boost_1_75_0
    colcon build --event-handler console_cohesion+ --merge-install

Try to run it:

    cd install
    lib\plotjuggler\plotjuggler.exe --no-splash

closed 2023-01-14 by facontidavide