PlotJuggler/plotjuggler-ros-plugins #28: Large 64bit integers are corrupted

EhrenbergerSoftware, 2021-09-14:
The ros1_parser can't handle large 64bit integers because it converts them to double. This corrupts values larger than 1<<53.

For example a uint64 value 2343141149457253134 in the ros bag is shown as 2343141149457253376 in the PlotJuggler.

The problematic line is here:
https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/17219af4f96391527209133bc093dec787b9a399/plugins/ros1_parsers/ros1_parser.cpp#L55

facontidavide, 2021-09-14:
This is a well-known problem with NO SOLUTION in the current implementation of PlotJuggler.

facontidavide, 2021-09-14:
note that the limiting factor is QWT itself, which supports only doubles. Solving this issue would be incredibly complex

jan-ehrenberger, 2021-09-14:
Thanks for the info. Yes, you are right. It seems that PlotJuggler uses doubles everywhere.
For my purposes I created a workaround that stores the values as strings, but I do understand it has some downsides. The stored values then can't be used for plotting.

```
diff --git a/plugins/ros1_parsers/ros1_parser.cpp b/plugins/ros1_parsers/ros1_parser.cpp
index e12390c..7477188 100644
--- a/plugins/ros1_parsers/ros1_parser.cpp
+++ b/plugins/ros1_parsers/ros1_parser.cpp
@@ -51,26 +51,26 @@ bool IntrospectionParser::parseMessage(MessageRef serialized_msg, double& timest
 
     if( it.second.getTypeID() ==  RosIntrospection::BuiltinType::UINT64 )
     {
-        uint64_t raw_value = it.second.extract<uint64_t>();
-        //      if( raw_value >= (1l<<53) ){
-        //        TODO: warn the user?
-        //      }
-        value = static_cast<double>(raw_value);
+      // TODO: Store the 64bit integer values properly.
+      uint64_t raw_value = it.second.extract<uint64_t>();
+      auto& series = _plot_data.getOrCreateStringSeries( key );
+      series.pushBack({ timestamp, std::to_string(raw_value)});
+      continue;
     }
     else if( it.second.getTypeID() ==  RosIntrospection::BuiltinType::INT64 )
     {
-        int64_t raw_value = it.second.extract<int64_t>();
-        //      if( raw_value >= (1l<<53) ){
-        //        TODO: warn the user?
-        //      }
-        value = static_cast<double>(raw_value);
+      // TODO: Store the 64bit integer values properly.
+      int64_t raw_value = it.second.extract<int64_t>();
+      auto& series = _plot_data.getOrCreateStringSeries( key );
+      series.pushBack({ timestamp, std::to_string(raw_value)});
+        continue;
     }
     else if( it.second.getTypeID() ==  RosIntrospection::BuiltinType::STRING ) {
-        // special case for strings
-        auto str = it.second.extract<std::string>();
-        auto& series = _plot_data.getOrCreateStringSeries( key );
-        series.pushBack({ timestamp, str});
-        continue;
+      // special case for strings
+      auto str = it.second.extract<std::string>();
+      auto& series = _plot_data.getOrCreateStringSeries( key );
+      series.pushBack({ timestamp, str});
+      continue;
     }
     else{
       value = it.second.convert<double>();
-- 
```

facontidavide, 2021-09-14:
Potentially a better solution (long term, not planning to do it in any way) is to modify the way internal storage is done to use Any everywhere and cast the value accordingly. The QWT visualization will still be wrong, but at least the number on the 2nd column of the left panel would be correct.
 
But this would be a massive refactoring that I am not planning to do on my own. If any company wants to SPONSOR this development, please get in touch.

closed 2021-09-14 by facontidavide