PlotJuggler/plotjuggler-ros-plugins #35: Not enough memory in the buffer stream when loading ROS2 data from CARLA simulator (Foxy Focal)

andrewbest-tri, 2021-12-13:
Greetings. I am running on ros2 foxy on ubuntu focal. 
I am running the latest version of plotjuggler from deb (sudo apt-get install ros-foxy-plotjuggler-ros).
I am attempting to load data from CARLA's object message which is an ObjectArray: 
    http://docs.ros.org/en/kinetic/api/derived_object_msgs/html/msg/Object.html

When I attempt to load my array, I receive the following error:
```
he plugin [DataLoad ROS2 bags] thrown the following exception: 

 Not enough memory in the buffer stream
```
I am not sure how to debug further, but I do have a significant amount of available RAM (somewhere around 100gb free).  

Any idea what may be causing this? If you need further information, I am happy to provide it.

facontidavide, 2021-12-14:
Investigating a little, it seems like this is originated in FastCDR, not plotjuggler.

You should search there for a solution.

closed 2022-02-11 by facontidavide

yuasabe, 2022-03-19:
@andrewbest-tri Hi, I was also having the same issue. Were you able to figure out a solution?

andrewbest-tri, 2022-03-21:
I have not. I have modified my input data to remove the channels that cause problems for the moment as a workaround.