PlotJuggler/plotjuggler-ros-plugins #37: The plugin [DataLoad ROS bags] thrown the following exception: Variant::extract -> wrong type

yvonl, 2022-01-21:
Hi,

I have a custom ROS message `VisualizationData` and a bagfile containing built-in ROS messages (like `std_msgs/Bool`, `sensor_msgs/Imu`, ...) and others custom ROS messages (including `VisualizationData`). When I load the bagfile, if I select the topic with Datatype `VisualizationData`, plotjuggler fails with error:

[image]

I don't have this issue with the others custom ROS messages.

OS: Ubuntu 20.04
Plotjuggler: from snap, PlotJuggler-3 3.0.7

`VisualizationData` is a composition of built-in ROS type/fixed-length and variable-length array and custom ROS messages :

`VisualizationData.msg`
```
Header header

int64 image_counter
bool zero_velocity_update

int8[] camera_id_array
FrameInfo[] frames
```

`FrameInfo.msg`
```
Point2D[] candidates
Landmark[] landmarks
LandmarkUncertainty[] landmark_uncertainties
NewLandmark[] new_landmarks
PatchInfo[] patches

float64 roll
float64 pitch
```

`Point2D.msg`
```
float32 x
float32 y
```

`Landmark.msg`
```
uint8 LANDMARK_FINAL = 0
uint8 LANDMARK_SURROUNDING_GOOD = 1
uint8 LANDMARK_SURROUNDING_BAD = 2
uint8 LANDMARK_NOT_CONVERGED = 3
uint8 LANDMARK_INTENSITY_ERROR = 4

uint8 status
Point2D point
```

`LandmarkUncertainty.msg`
```
Point2D point
uint32 number

float64 sigma1
float64 sigma2
float64 sigma_angle
```

`NewLandmark.msg`
```
uint8 NEW_LANDMARK_CANDIDATE = 0
uint8 NEW_LANDMARK_CONVERGED_VALID = 1
uint8 NEW_LANDMARK_CONVERGED_REJECTED = 2
uint8 NEW_LANDMARK_NOT_CONVERGED = 3

uint8 status
Point2D point
uint32 number
```

`PatchInfo.msg`
```
uint8 PATCH_SUCCESS = 0
uint8 PATCH_NOT_IN_FRAME = 1
uint8 PATCH_OUTLIER = 2
uint8 PATCH_NOT_CONVERGED = 3
uint8 PATCH_TOO_LARGE_INTENSITY_ERROR = 4

uint8 status
Point2D point
Point2D[4] corners
float64 data
```

I tracked down the issue to [ros1_parser.cpp: IntrospectionParser::parseMessage](https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/development/plugins/ros1_parsers/ros1_parser.cpp#L52-L77) and the functions raising the exception to [variant.hpp: Variant::extract](https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/development/plugins/ros1_introspection/include/ros_type_introspection/utils/variant.hpp#L169-L204). But I don't know why it fails to parse the message, given its definition.

Any idea what may be causing this?

facontidavide, 2022-02-11:
can you give me a rosbag to replicate the error?

yvonl, 2022-02-15:
Yes, of course. With this bagfile [visualizationData-wrong_type.zip](https://github.com/PlotJuggler/plotjuggler-ros-plugins/files/8068999/visualizationData-wrong_type.zip) you should be able to reproduce the error.