PlotJuggler/plotjuggler-ros-plugins #38: ros topic subscriber plugin not

hoefes, 2022-01-21:
I installed the ros package version via
`sudo apt install ros-noetic-plotjuggler`
and launched the application via 
`rosrun plotjuggler plotjuggler`

under the streaming dropdown-menu, I can choose "UDP Server", "WebSocket Server" and "ZMQ Subscriber". I am missing the option "ROS Topic Subscriber". Is this the desired behavior? How can I get the ros topic subscriber?
[image]
?

facontidavide, 2022-01-23:
can you provide the console output?

hoefes, 2022-01-23:
There you go:
```
wmds@wmds-lidar:~$ rosrun plotjuggler plotjuggler
Loading compatible plugins from directory:  "/opt/ros/noetic/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a MessageParser plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a MessageParser plugin"
Number of plugins loaded:  8 

Loading compatible plugins from directory:  "/opt/ros/noetic/lib/plotjuggler_ros"
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/home/wmds/catkin_ws/devel/lib/plotjuggler_ros"
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/home/wmds/.local/share/PlotJuggler"
Number of plugins loaded:  0
```
Thanks for looking into this!

facontidavide, 2022-01-23:
sudo apt install ros-noetic-plotjuggler-ros

closed 2022-01-23 by facontidavide

hoefes, 2022-01-23:
thanks a lot, maybe consider adding this line to the repo's README.