PlotJuggler/plotjuggler-ros-plugins #44: Newest version of ros-plugins cannot be built. Missing ros2bag_transport package

bednarhonza, 2022-03-02:
Hello,
I wanted the newest version of ROS2 plugin, but it is still not available as a .deb package through apt system. I tried to build it in my own colcon workspace, but it misses the ros2bag_transport package. I have this package installed, so I do not see the problem. Could you help me with that? Or do you know when there will be a .deb package version online?

````
Finished` <<< plotjuggler [1min 31s]
Starting >>> plotjuggler_ros
--- stderr: plotjuggler_ros
In file included from /home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataStreamROS2/datastream_ros2.h:11,
                 from /home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataStreamROS2/datastream_ros2.cpp:1:
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/ros2_parsers/generic_subscription.hpp:26:10: fatal error: rosbag2_transport/logging.hpp: No such file or directory
   26 | #include "rosbag2_transport/logging.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [plugins/CMakeFiles/DataStreamROS2.dir/build.make:76: plugins/CMakeFiles/DataStreamROS2.dir/DataStreamROS2/datastream_ros2.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataStreamROS2/datastream_ros2.h:11,
                 from /home/honzabednar/plotjuggler_workspace/build/plotjuggler_ros/plugins/DataStreamROS2_autogen/JX3IUUCCPQ/moc_datastream_ros2.cpp:9,
                 from /home/honzabednar/plotjuggler_workspace/build/plotjuggler_ros/plugins/DataStreamROS2_autogen/mocs_compilation.cpp:2:
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/ros2_parsers/generic_subscription.hpp:26:10: fatal error: rosbag2_transport/logging.hpp: No such file or directory
   26 | #include "rosbag2_transport/logging.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [plugins/CMakeFiles/DataStreamROS2.dir/build.make:63: plugins/CMakeFiles/DataStreamROS2.dir/DataStreamROS2_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: plugins/CMakeFiles/DataStreamROS2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataLoadROS2/dataload_ros2.cpp: In member function ‘virtual bool DataLoadROS2::readDataFromFile(PJ::FileLoadInfo*, PJ::PlotDataMapRef&)’:
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataLoadROS2/dataload_ros2.cpp:53:31: warning: ‘using StorageOptions = struct rosbag2_storage::StorageOptions’ is deprecated: use rosbag2_storage::StorageOptions instead [-Wdeprecated-declarations]
   53 |   rosbag2_cpp::StorageOptions storageOptions;
      |                               ^~~~~~~~~~~~~~
In file included from /home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataLoadROS2/dataload_ros2.cpp:21:
/opt/ros/galactic/include/rosbag2_cpp/storage_options.hpp:23:7: note: declared here
   23 | using StorageOptions [[deprecated("use rosbag2_storage::StorageOptions instead")]] =
      |       ^~~~~~~~~~~~~~
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/TopicPublisherROS2/publisher_ros2.cpp: In constructor ‘TopicPublisherROS2::TopicPublisherROS2()’:
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/TopicPublisherROS2/publisher_ros2.cpp:28:28: error: ‘rclcpp::executor’ has not been declared
   28 |   auto exec_args = rclcpp::executor::ExecutorArgs();
      |                            ^~~~~~~~
/home/honzabednar/plotjuggler_workspace/src/plotjuggler-ros-plugins/plugins/DataLoadROS2/dataload_ros2.cpp:153:18: error: ‘using element_type = class rosbag2_cpp::readers::SequentialReader’ {aka ‘class rosbag2_cpp::readers::SequentialReader’} has no member named ‘reset’
  153 |     _bag_reader->reset();
      |                  ^~~~~
make[2]: *** [plugins/CMakeFiles/DataLoadROS2.dir/build.make:76: plugins/CMakeFiles/DataLoadROS2.dir/DataLoadROS2/dataload_ros2.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:217: plugins/CMakeFiles/DataLoadROS2.dir/all] Error 2
make[2]: *** [plugins/CMakeFiles/TopicPublisherROS2.dir/build.make:76: plugins/CMakeFiles/TopicPublisherROS2.dir/TopicPublisherROS2/publisher_ros2.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:161: plugins/CMakeFiles/TopicPublisherROS2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< plotjuggler_ros [23.9s, exited with code 2]

Summary: 2 packages finished [1min 55s]
  1 package failed: plotjuggler_ros
  2 packages had stderr output: plotjuggler plotjuggler_ros
```

facontidavide, 2022-03-02:
ROS distro and branch?

bednarhonza, 2022-03-03:
ROS Galactic - newest .deb packages
plotjuggler-ros-plugins - development branch, commit: 1a6dc358

Thanks!

facontidavide, 2022-03-03:
change to **rolling** branch

bednarhonza, 2022-03-03:
I found the issue. The `rosbag2_transport` package has `qos.hpp` file in the `src` folder, not in the `include` folder.
So the `qos.hpp` header file is missing in the `opt/ros/galactic/include/rosbag2_transport/` folder. I manually copied the `qos.hpp` file in the folder `opt/ros/galactic/include/rosbag2_transport/` and it works now.

Thanks for the help!