PlotJuggler/plotjuggler-ros-plugins #49: How to install specific (older) version of plotjuggler and needed plugins?

krtt, 2022-04-18:
I have different systems. One is Ubuntu 18.04, another one is Kubuntu 18.04. Currently I use Plottjuggler 3.4 but it is too memory hungry for analyzing ros bags. An older version of Plotjuggler 3.0.7 was better for this. I installed the v3.0.7 but I fail to install the plotjuggler_ros package (Unable to locate package plotjuggler_ros). What is the fastest and/or the cleanest way?

facontidavide, 2022-04-18:
No way, but I am happy to analyze why it is consuming more memory, since it is not supposed to. Can you share the rosbag, please?

krtt, 2022-04-18:
Cannot share publicly. But can describe:
 - bags are usually in size 5-200 GB
 - a small bag of for example 5-6 GB has at least 400000 msgs
 - number of topics can be reduced for analyzing a very specific problem but it is not comfortable if analyzing yet unknown problem where need to load all topics before narrowing it down to a specific set of topics
 - some example messages are from lidars, radars, cameras.
 
Maybe you already have some large ros bag that you can compare between 3.0.7 and 3.4. They seem to handle the topics differently.

If it's not helping, let me know what kind of info exactly is needed and I will see what we can do. Thanks for your help.

facontidavide, 2023-12-21:
Closing this for one of the following reasons:

- absolate
- you didn't provide an easy way to reproduce it
- You should test latest version (PlotJuggler 3.8.4 + ROS plugins 2.0.0)

closed 2023-12-21 by facontidavide