PlotJuggler/plotjuggler-ros-plugins #59: plotjugger_ros build failures on Rolling

nuclearsandwich, 2023-01-04:
ROS distro: rolling
OS: Ubuntu 22.04 (Jammy)
plotjuggler_ros version: 1.5.1-2

Example failing build: https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__plotjuggler_ros__ubuntu_jammy_amd64__binary/72/

Console log excerpt:
```
00:05:55.155 cd /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins && /usr/lib/ccache/c++ -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DDISABLE_RULE_EDITING -DFMT_HEADER_ONLY -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_PLUGIN -DQT_SHARED -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DQWT_MOC_INCLUDE -DRCUTILS_ENABLE_FAULT_INJECTION -DTopicPublisherROS2_EXPORTS -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/plugins -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/include -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/3rdparty -I/usr/include/x86_64-linux-gnu/qt5/QtWebSockets -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/plugins/. -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/plugins/ros2_introspection/include -I/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/plugins/ros2_parsers -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtConcurrent -isystem /usr/include/x86_64-linux-gnu/qt5/QtXml -isystem /usr/include/x86_64-linux-gnu/qt5/QtSvg -isystem /opt/ros/rolling/include/rclcpp -isystem /opt/ros/rolling/include/rcpputils -isystem /opt/ros/rolling/include/rosbag2_transport -isystem /opt/ros/rolling/include/sensor_msgs -isystem /opt/ros/rolling/include/geometry_msgs -isystem /opt/ros/rolling/include/nav_msgs -isystem /opt/ros/rolling/include/diagnostic_msgs -isystem /opt/ros/rolling/include/plotjuggler_msgs -isystem /opt/ros/rolling/include/tf2_msgs -isystem /opt/ros/rolling/include/tf2_ros -isystem /opt/ros/rolling/include -isystem /opt/ros/rolling/include/fastcdr -isystem /opt/ros/rolling/include/ament_index_cpp -isystem /opt/ros/rolling/include/libstatistics_collector -isystem /opt/ros/rolling/include/builtin_interfaces -isystem /opt/ros/rolling/include/rosidl_runtime_c -isystem /opt/ros/rolling/include/rcutils -isystem /opt/ros/rolling/include/rosidl_typesupport_interface -isystem /opt/ros/rolling/include/rosidl_runtime_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/rolling/include/rmw -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/rolling/include/rcl -isystem /opt/ros/rolling/include/rcl_interfaces -isystem /opt/ros/rolling/include/rcl_logging_interface -isystem /opt/ros/rolling/include/rcl_yaml_param_parser -isystem /opt/ros/rolling/include/libyaml_vendor -isystem /opt/ros/rolling/include/tracetools -isystem /opt/ros/rolling/include/statistics_msgs -isystem /opt/ros/rolling/include/rosgraph_msgs -isystem /opt/ros/rolling/include/rosidl_typesupport_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_c -isystem /opt/ros/rolling/include/keyboard_handler -isystem /opt/ros/rolling/include/rosbag2_compression -isystem /opt/ros/rolling/include/pluginlib -isystem /opt/ros/rolling/include/class_loader -isystem /opt/ros/rolling/include/rosbag2_cpp -isystem /opt/ros/rolling/include/rosbag2_storage -isystem /opt/ros/rolling/include/rosbag2_interfaces -isystem /opt/ros/rolling/include/moodycamel -isystem /opt/ros/rolling/include/std_msgs -isystem /opt/ros/rolling/include/action_msgs -isystem /opt/ros/rolling/include/unique_identifier_msgs -isystem /opt/ros/rolling/include/message_filters -isystem /opt/ros/rolling/include/rclcpp_action -isystem /opt/ros/rolling/include/rcl_action -isystem /opt/ros/rolling/include/tf2 -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -g -fno-omit-frame-pointer  -fPIC -fPIC -fPIC -std=gnu++17 -MD -MT plugins/CMakeFiles/TopicPublisherROS2.dir/TopicPublisherROS2_autogen/mocs_compilation.cpp.o -MF CMakeFiles/TopicPublisherROS2.dir/TopicPublisherROS2_autogen/mocs_compilation.cpp.o.d -o CMakeFiles/TopicPublisherROS2.dir/TopicPublisherROS2_autogen/mocs_compilation.cpp.o -c /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/mocs_compilation.cpp
00:05:55.156 In file included from /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/TH6L5D5FXO/../../../../plugins/TopicPublisherROS2/publisher_ros2.h:18,
00:06:02.202                  from /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/TH6L5D5FXO/moc_publisher_ros2.cpp:10,
00:06:02.202                  from /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/mocs_compilation.cpp:2:
00:06:02.202 /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/TH6L5D5FXO/../../../../plugins/TopicPublisherROS2/generic_publisher.h: In constructor ‘GenericPublisher::GenericPublisher(rclcpp::node_interfaces::NodeBaseInterface*, const string&, const rosidl_message_type_support_t&)’:
00:06:02.202 /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/TH6L5D5FXO/../../../../plugins/TopicPublisherROS2/generic_publisher.h:28:101: error: no matching function for call to ‘rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*&, const string&, const rosidl_message_type_support_t&, rcl_publisher_options_t)’
00:06:02.202    28 |     : rclcpp::PublisherBase(node_base, topic_name, type_support, rcl_publisher_get_default_options())
00:06:02.202       |                                                                                                     ^
00:06:02.202 In file included from /opt/ros/rolling/include/rclcpp/rclcpp/callback_group.hpp:27,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/any_executable.hpp:20,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/memory_strategy.hpp:25,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/memory_strategies.hpp:18,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/executor_options.hpp:20,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/executor.hpp:37,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/executors.hpp:21,
00:06:02.202                  from /opt/ros/rolling/include/rclcpp/rclcpp/rclcpp.hpp:155,
00:06:02.202                  from /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/TH6L5D5FXO/../../../../plugins/TopicPublisherROS2/publisher_ros2.h:9,
00:06:02.202                  from /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/TH6L5D5FXO/moc_publisher_ros2.cpp:10,
00:06:02.202                  from /tmp/binarydeb/ros-rolling-plotjuggler-ros-1.5.1/obj-x86_64-linux-gnu/plugins/TopicPublisherROS2_autogen/mocs_compilation.cpp:2:
00:06:02.202 /opt/ros/rolling/include/rclcpp/rclcpp/publisher_base.hpp:77:3: note: candidate: ‘rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, const string&, const rosidl_message_type_support_t&, const rcl_publisher_options_t&, const rclcpp::PublisherEventCallbacks&, bool)’
00:06:02.202    77 |   PublisherBase(
00:06:02.202       |   ^~~~~~~~~~~~~
00:06:02.202 /opt/ros/rolling/include/rclcpp/rclcpp/publisher_base.hpp:77:3: note:   candidate expects 6 arguments, 4 provided
00:06:02.202 /opt/ros/rolling/include/rclcpp/rclcpp/publisher_base.hpp:60:7: note: candidate: ‘rclcpp::PublisherBase::PublisherBase(const rclcpp::PublisherBase&)’
00:06:02.202    60 | class PublisherBase : public std::enable_shared_from_this<PublisherBase>
00:06:02.202       |       ^~~~~~~~~~~~~
00:06:02.202 /opt/ros/rolling/include/rclcpp/rclcpp/publisher_base.hpp:60:7: note:   candidate expects 1 argument, 4 provided
```

hodnajit, 2023-01-10:
I created a pull request, which works for me:
https://github.com/PlotJuggler/plotjuggler-ros-plugins/pull/60

texhnolyze, 2023-02-18:
Looking at the [jenkins build](https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__plotjuggler_ros__ubuntu_jammy_amd64__binary/) it seems the build works again.
@nuclearsandwich can this be closed?

closed 2023-02-20 by facontidavide