PlotJuggler/plotjuggler-ros-plugins #62: Segmentation Fault by loading Layout with tab separations

PSotiriadis, 2023-02-06:
# Problem description
Receive "Segmentation fault" trying to load saved layout, which contains vertical or horizontal tab separations. Launching plotjuggler and loading layout without tab separations, is working without problems. 

# Steps to reproduce (important)
**- Describe your platform / Operative System.**
  Ubuntu 22.04 with ROS2 Humble.

**- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".**
  Yes, I also receive "Segmentation Fault" Error! 
  In details, I start plotjuggler using "ros2 run plotjuggler plotjuggler -t" and I save the following layout under name jugplot.xml
[image]
I stop plotjuggler and call it again with "ros2 run plotjuggler plotjuggler". Then from 'File->Layout->load Data and Window plots layout' I try to load my jugplot.xml layout and I receive:
[image]

RCMast3r, 2023-03-01:
I am having the same issue

facontidavide, 2023-03-01:
I hope to have time investigating this during the weekend

federicociresola, 2023-07-04:
Hi,
I have the same problem with the same setup ROS2-Humble on Ubuntu 22.04. 
I am using version 3.7.1 which I downloaded from the official ROS repositories.

alexbuyval, 2023-07-07:
Hi,

I have the same issue too. 
[image]

facontidavide, 2023-07-07:
To all of you folks. Can you PLEASE give me something that I can run on my computer to reproduce this?

Preferably a rosbag + the incriminated layout XML

alexbuyval, 2023-07-07:
@facontidavide I hope it will be helpful.
However almost all messages are not standard in bag file. If it is critical let me know and I'll prepare another bag file.
[rosbag2_2023_07_07-13_52_21.zip](https://github.com/PlotJuggler/plotjuggler-ros-plugins/files/11985350/rosbag2_2023_07_07-13_52_21.zip)
[plot_mppi_resid.zip](https://github.com/PlotJuggler/plotjuggler-ros-plugins/files/11985353/plot_mppi_resid.zip)

scrice, 2023-10-27:
Still experiencing this issue on Windows 10 wsl2 Ubuntu 22.04 Humble.
Version 3.7.1 installed via "sudo apt install ros-humble-plotjuggler-ros"

DavideLanzoniSacmi, 2023-11-29:
> # Problem description
> Receive "Segmentation fault" trying to load saved layout, which contains vertical or horizontal tab separations. Launching plotjuggler and loading layout without tab [...]

Same exact error

azurechen1203, 2023-12-21:
@facontidavide: Having the same issue as well. The segmentation fault showed up even without data.

## Steps to reproduce:

**Platofrm/operative system:** Ubuntu 22.04 with ROS2 Humble

**Steps:** 
1. ros2 run plotjuggler plotjuggler
2.  right click -> split vertically
3. click save window/plot layout
4. close tab1
5. on tab2, load the .xml file

[Screencast from 21-12-23 11:25:28.webm](https://github.com/PlotJuggler/plotjuggler-ros-plugins/assets/141415841/4a6209a1-9215-48f2-9841-2f413850ac24)

**Results:**
Plotjuggler unexpectedly closed with the following error message:

[image]

**Note: **
The error doesn't happen if not splitting the plot.

closed 2023-12-21 by facontidavide

azurechen1203, 2024-01-04:
Sorry, how could I update it to 3.8.4?

azurechen1203, 2024-01-04:
Yeah! Already tried that but it's still 3.8.1

[image]

facontidavide, 2024-01-04:
which version of ROS are you using?
You must wait for the release; I am afraid

PSotiriadis, 2024-01-10:
I updated my system today, plotjuggler is updated to V3.8.5 and the problem no longer exists. Thank you @facontidavide 👍