PlotJuggler/plotjuggler-ros-plugins #63: std::bad_alloc on ROS2 Bag load

AntoineHX, 2023-02-28:
Hi,

I was trying to load a simple ROS2 bag with the load data from file (by selecting the metadata.yaml of the ROSbag) but it always throw the same error (std::bad_alloc) while loading the data.
I'm using [plotjuggler latest version](https://github.com/facontidavide/PlotJuggler/commit/b53ea82d8a6bd27a42879b9907c9ab5a8de6e224) with [plotjuggler-ros-plugins v1.7.3](https://github.com/PlotJuggler/plotjuggler-ros-plugins/commit/ccd0447bca5ca7b158e0d6e238d47b8b584d8e23). Is this an expected behavior ?

The metadata of the simple ROSBag :
```YAML
rosbag2_bagfile_information:
  version: 4
  storage_identifier: sqlite3
  relative_file_paths:
    - rosbag2_2023_02_28-11_44_54_0.db3
  duration:
    nanoseconds: 6430324604
  starting_time:
    nanoseconds_since_epoch: 1677581094320974926
  message_count: 6433
  topics_with_message_count:
    - topic_metadata:
        name: /wrench_ee
        type: geometry_msgs/msg/WrenchStamped
        serialization_format: cdr
        offered_qos_profiles: "- history: 3\n  depth: 0\n  reliability: 1\n  durability: 2\n  deadline:\n    sec: 2147483647\n    nsec: 4294967295\n  lifespan:\n    sec: 2147483647\n    nsec: 4294967295\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 2147483647\n    nsec: 4294967295\n  avoid_ros_namespace_conventions: false"
      message_count: 6433
  compression_format: ""
  compression_mode: ""
```
The stack trace :
```bash
double free or corruption (!prev)
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e8c70ef, in QApplication::notify(QObject*, QEvent*)
#30   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e8bda65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#29   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e91f1eb, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#28   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e91c35c, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#27   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e8c6456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f731dea5809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#25   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e8c7342, in QApplication::notify(QObject*, QEvent*)
#24   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e8bda65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#23   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9002b5, in QWidget::event(QEvent*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9b4034, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9b3e72, in QAbstractButton::setIconSize(QSize const&)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9b2a2d, in QAbstractButton::toggled(bool)
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9b2805, in QAbstractButton::clicked(bool)
#18   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f731ded128b, in QMetaObject::activate(QObject*, int, int, void**)
#17   Source "/home/antoineh/Plotjuggler_ws/build/plotjuggler/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 514, in MainWindow::qt_metacall(QMetaObject::Call, int, void**) [0x56083cf09aa9]
        511:         return _id;
        512:     if (_c == QMetaObject::InvokeMetaMethod) {
        513:         if (_id < 66)
      > 514:             qt_static_metacall(this, _c, _id, _a);
        515:         _id -= 66;
        516:     } else if (_c == QMetaObject::RegisterMethodArgumentMetaType) {
        517:         if (_id < 66)
#16   Source "/home/antoineh/Plotjuggler_ws/build/plotjuggler/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 422, in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) [0x56083cf07d48]
        419:         case 61: _t->on_buttonRecentLayout_clicked(); break;
        420:         case 62: _t->on_pushButtonLoadLayout_clicked(); break;
        421:         case 63: _t->on_pushButtonSaveLayout_clicked(); break;
      > 422:         case 64: _t->on_pushButtonLoadDatafile_clicked(); break;
        423:         case 65: _t->on_actionColorMap_Editor_triggered(); break;
        424:         default: ;
        425:         }
#15   Source "/home/antoineh/Plotjuggler_ws/src/PlotJuggler/plotjuggler_app/mainwindow.cpp", line 2987, in MainWindow::on_pushButtonLoadDatafile_clicked() [0x56083cf5acd5]
       2984:   directory_path = QFileInfo(fileNames[0]).absolutePath();
       2985:   settings.setValue("MainWindow.lastDatafileDirectory", directory_path);
       2986: 
      >2987:   if (loadDataFromFiles(fileNames))
       2988:   {
       2989:     updateRecentDataMenu(fileNames);
       2990:   }
#14   Source "/home/antoineh/Plotjuggler_ws/src/PlotJuggler/plotjuggler_app/mainwindow.cpp", line 1368, in MainWindow::loadDataFromFiles(QStringList) [0x56083cf56073]
       1365:     {
       1366:       info.prefix = filename_prefix[info.filename];
       1367:     }
      >1368:     auto added_names = loadDataFromFile(info);
       1369:     if (!added_names.empty())
       1370:     {
       1371:       loaded_filenames.push_back(filenames[i]);
#13   Source "/home/antoineh/Plotjuggler_ws/src/PlotJuggler/plotjuggler_app/mainwindow.cpp", line 1534, in MainWindow::loadDataFromFile[abi:cxx11](PJ::FileLoadInfo const&) [0x56083ceef419]
       1531:     }
       1532:     catch (std::exception& ex)
       1533:     {
      >1534:       QMessageBox::warning(this, tr("Exception from the plugin"),
       1535:                            tr("The plugin [%1] thrown the following exception: \n\n %3\n")
       1536:                                .arg(dataloader->name())
       1537:                                .arg(ex.what()));
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731eafa478, in QMessageBox::critical(QWidget*, QString const&, QString const&, int, int, int)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e8fbc28, in QWidget::~QWidget()
#10   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f731deceeed, in QObjectPrivate::deleteChildren()
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9f935c, in QLabel::~QLabel()
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9f9328, in QLabel::~QLabel()
#7    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731e9f90ce, in QLabel::setBuddy(QWidget*)
#6    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f731eaa81bc, in QWidgetTextControl::~QWidgetTextControl()
#5    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f731ded9506, in QObject::~QObject()
BFD: DWARF error: section .debug_info is larger than its filesize! (0x93ef57 vs 0x530ea0)
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f731d752fab, in pthread_attr_setschedparam
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f731d7512fb, in pthread_attr_setschedparam
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f731d74926d, in __fsetlocking
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f731d6de858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f731d6ff00b, in gsignal
```
Cheers

j953302, 2023-06-19:
Hello, I met this probelm, too. Had you fugure out this problem?

j953302, 2023-06-20:
Yeah... Now I use it in the same way. Really cumbersome. Even I reinstall it still fail....

facontidavide, 2023-11-27:
can you tell me if the **snap** version is affected?

AntoineHX, 2023-12-07:
The snap version seems to work fine for built-in messages. But I didn't manage to use custom messages with it : 
```
The plugin thrown the following exception: 

 package 'su_dsg_interfaces' not found, searching: [/snap/plotjuggler/100/opt/ros/foxy]
```
It seems it doesn't use sourced interfaces like the Debian version.

To sum-up, from what I've experienced:

- Snap : Loading ROS2 bag working. Custom messages types not supported (Subscribing or bag loading): 
[custom_record.zip](https://github.com/PlotJuggler/plotjuggler-ros-plugins/files/13599889/custom_record.zip)

- Debian: Subscribing with custom messages types working. Any bag loading makes it crash: [simple_record.zip](https://github.com/PlotJuggler/plotjuggler-ros-plugins/files/13599792/simple_record.zip)