PlotJuggler/plotjuggler-ros-plugins #65: [Humble] Segmentation fault (Address not mapped to object [0x1e0]

Ogunniran, 2023-05-23:
* Building and running the packages works just fine:
   * PlotJuggler --> main branch
   * plotjuggler-ros-plugins --> Humble branch
   * plotjuggler_msgs --> main branch
* However after selecting a ros2 message (of sensor_msgs/msg/JointState type) the error below occured:  

```
Loading compatible plugins from directory:  "/home/ws_ros2/install/plotjuggler_ros/lib/plotjuggler_ros"
"libDataLoadROS2.so is a DataLoader plugin"
"libDataStreamROS2.so is a DataStreamer plugin"
[WARN] [1684830914.399383185] [rclcpp]: logging was initialized more than once
"libTopicPublisherROS2.so is a StatePublisher plugin"
Number of plugins loaded:  3 

Loading compatible plugins from directory:  "/home/ws_ros2/install/plotjuggler/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserDataTamer.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  14 

Loading compatible plugins from directory:  "/home/paul/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f4d8d4b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e16c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e1cbfd4, in 
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e1c8d3f, in 
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e172e46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f4d8d4b9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e174363, in QApplication::notify(QObject*, QEvent*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e16c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e1af4ed, in QWidget::event(QEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e2641e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e263fc3, in 
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e26239d, in 
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f4d8e262115, in QAbstractButton::clicked(bool)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f4d8d4f14e4, in 
#17   Source "/home/paul/ws_ros2/build/plotjuggler/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_mainwindow.cpp", line 515, in qt_metacall [0x55614cd2e199]
        512:         return _id;
        513:     if (_c == QMetaObject::InvokeMetaMethod) {
        514:         if (_id < 66)
      > 515:             qt_static_metacall(this, _c, _id, _a);
        516:         _id -= 66;
        517:     } else if (_c == QMetaObject::RegisterMethodArgumentMetaType) {
        518:         if (_id < 66)
#16   Source "/home/paul/ws_ros2/src/PlotJuggler/plotjuggler_app/mainwindow.cpp", line 3349, in on_buttonStreamingStart_clicked [0x55614cd70fd2]
       3346:   ui->buttonStreamingStart->setEnabled(false);
       3347:   if (ui->buttonStreamingStart->text() == "Start")
       3348:   {
      >3349:     startStreamingPlugin(ui->comboStreaming->currentText());
       3350:   }
       3351:   else
       3352:   {
#15   Source "/home/paul/ws_ros2/src/PlotJuggler/plotjuggler_app/mainwindow.cpp", line 1695, in startStreamingPlugin [0x55614cd6788b]
       1692:   try
[... 172 lines cut ...]
    |   139:   {
    | > 140:     it.second.setMaximumRangeX(range);
    |   141:   }
    |   142:   for (auto& it : user_defined)
    | Source "/home/paul/ws_ros2/src/PlotJuggler/plotjuggler_base/include/PlotJuggler/timeseries.h", line 45, in trimRange
    |    43:   {
    |    44:     _max_range_x = max_range;
    | >  45:     trimRange();
    |    46:   }
    | Source "/home/paul/ws_ros2/src/PlotJuggler/plotjuggler_base/include/PlotJuggler/timeseries.h", line 86, in trimRange
    |    85: private:
    | >  86:   void trimRange()
    |    87:   {
    |    88:     if(_max_range_x < std::numeric_limits<double>::max())
      Source "/home/paul/ws_ros2/src/PlotJuggler/plotjuggler_base/include/PlotJuggler/timeseries.h", line 90, in setMaximumRangeX [0x56461db35c45]
         87:   {
         88:     if(_max_range_x < std::numeric_limits<double>::max())
         89:     {
      >  90:       auto const back_point_x = _points.back().x;
         91:       while (_points.size() > 2 && (back_point_x - _points.front().x) > _max_range_x)
         92:       {
         93:         this->popFront();
Segmentation fault (Address not mapped to object [0x1e0])
[ros2run]: Segmentation fault
```

facontidavide, 2023-05-23:
I just pushed a fix in the main branch of PlotJuggler. Check if your problem is solved, please

Ogunniran, 2023-05-23:
Thanks for the quick fix. Tested it and it works fine now.

closed 2023-05-23 by Ogunniran