PlotJuggler/plotjuggler-ros-plugins #71: ROS 2 release

Noel215, 2023-07-17:
Hi,

I created this [Pull Request](https://github.com/PlotJuggler/plotjuggler-ros-plugins/pull/66) to enable pal_statistics plugin for ROS 2 and it got merged.

Could you release it for humble and rolling in order to be able to use that feature without having to clone the repository in my workspace?

Thanks.

facontidavide, 2023-07-17:
You broke compilation in ROS1.
Would you please find a solution?

Noel215, 2023-07-19:
`pal_statistics` was not released for ROS Noetic. 
Requested in https://github.com/ros/rosdistro/pull/38039 for noetic and in https://github.com/ros/rosdistro/pull/38038 for rolling

Noel215, 2023-09-12:
Friendly reminder @facontidavide. 
I released `pal_statistics` for `noetic` and `rolling`, but I don't know if that was enough to fix the issue and release it for ROS 2 then.

facontidavide, 2024-02-02:
not necessary anymore since version 2.0

closed 2024-02-02 by facontidavide