PlotJuggler/plotjuggler-ros-plugins #75: Change `strings_chain` type from `static_vector to `std::vector` in `RosIntrospection::CreateStringFromTreeLeaf()`

m2themax, 2023-12-13:
Can we change the type of `strings_chain` from `boost::container::static_vector` with fixed length of 16 to `std::vector` in `RosIntrospection::CreateStringFromTreeLeaf()`, i.e. here

https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/e629cd6a1244b4c14402272ea1c04977a9a1b136/plugins/ros1_introspection/src/stringtree_leaf.cpp#L48

The fixed size causes bad_alloc error when loading bags with message depth above 16. With message depth I mean the depth of the tree representing the message type, i.e. the tree represented by `leaf` in this function.

m2themax, 2023-12-13:
Created PR: https://github.com/PlotJuggler/plotjuggler-ros-plugins/pull/76

facontidavide, 2023-12-13:
I am refactoring the library completely. Check branch "main".

The ROS parser have been moved here: https://github.com/facontidavide/PlotJuggler/tree/ros_refactored/plotjuggler_plugins/ParserROS

closed 2023-12-13 by facontidavide

m2themax, 2023-12-13:
> I am refactoring the library completely. Check branch "main".
> 
> The ROS parser have been moved here: https://github.com/facontidavide/PlotJuggler/tree/ros_refactored/plotjuggler_plugins/ParserROS

The bad_alloc problem is fixed on your branch 🥳 

Can you estimate when this branch will be merged to main and when the updated version will be available in the apt mirror?

facontidavide, 2023-12-14:
done. I have no idea when the APT will be available, it is outside my control