PlotJuggler/plotjuggler-ros-plugins #81: Proper plotjuggler branch to pair with plotjuggler-ros-plugins' humble branch

d-loret, 2024-02-06:
We were using the `PlotJuggler/plotjuggler-ros-plugins`'s `humble` branch together with `facontidavide/PlotJuggler`'s `main` branch in the past, but recently we started to have build errors:
```
/opt/casah/walking_arm/v0.0.6/src/plotjuggler_ros_plugins/plugins/DataLoadROS2/dataload_ros2.cpp: In member function ‘virtual bool DataLoadROS2::readDataFromFile(PJ::FileLoadInfo*, PJ::PlotDataMapRef&)’:
/opt/casah/walking_arm/v0.0.6/src/plotjuggler_ros_plugins/plugins/DataLoadROS2/dataload_ros2.cpp:130:14: error: ‘struct PJ::FileLoadInfo’ has no member named ‘selected_datasources’
  130 |   if (!info->selected_datasources.empty())
      |              ^~~~~~~~~~~~~~~~~~~~
/opt/casah/walking_arm/v0.0.6/src/plotjuggler_ros_plugins/plugins/DataLoadROS2/dataload_ros2.cpp:132:28: error: ‘struct PJ::FileLoadInfo’ has no member named ‘selected_datasources’
  132 |     _config.topics = info->selected_datasources;
      |                            ^~~~~~~~~~~~~~~~~~~~
/opt/casah/walking_arm/v0.0.6/src/plotjuggler_ros_plugins/plugins/DataLoadROS2/dataload_ros2.cpp:241:9: error: ‘struct PJ::FileLoadInfo’ has no member named ‘selected_datasources’
  241 |   info->selected_datasources = _config.topics;
      |         ^~~~~~~~~~~~~~~~~~~~

```
Is there a specific branch of `facontidavide/PlogJuggler` that we should be using with `PlotJuggler/plotjuggler-ros-plugins'` humble branch?

facontidavide, 2024-02-06:
No, you should use the main branch

salmagro, 2024-02-19:
I am facing also an error in ros2_humbele when using both in main:
```
pkgs/build/plotjuggler_ros/src/TopicPublisherROS2_autogen/TH6L5D5FXO/../../../../../src/ws_plotjuggler/plotjuggler-ros-plugins/src/TopicPublisherROS2/generic_publisher.h:32:117: error: no matching function for call to ‘rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*&, const string&, const rosidl_message_type_support_t&, rcl_publisher_options_t, rclcpp::PublisherEventCallbacks&, bool)’
```
With the branches:
```
ws_plotjuggler/PlotJuggler(main)$ git log
commit 2376fdf75b4e01da5d964eb9ed2260f3ad025dca (HEAD -> main, origin/main, origin/HEAD)
Author: Davide Faconti <davide.faconti@gmail.com>
Date:   Wed Feb 14 21:46:26 2024 +0100

    fix #935 and #934
```
and
```
ws_plotjuggler/plotjuggler-ros-plugins(main)$ git log
commit 5932e98f0c3e022b2c616750302d3b1bba0a4ad1 (HEAD -> main, tag: 2.1.0, origin/main)
Author: Davide Faconti <davide.faconti@gmail.com>
Date:   Sun Feb 4 16:25:48 2024 +0100

    2.1.0
```

facontidavide, 2024-02-19:
@salmagro same error or other?

facontidavide, 2024-02-19:
I just tested and it works just fine. There must be some version mismatch on your side

Tiqo01, 2024-02-26:
I have the same problem ros humble (source build) 

`ws_plotjuggler/src/plotjuggler-ros-plugins/src/TopicPublisherROS2/generic_publisher.h:32:117: error: no matching function for call to ‘rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*&, const string&, const rosidl_message_type_support_t&, rcl_publisher_options_t, rclcpp::PublisherEventCallbacks&, bool)’
   32 |   : rclcpp::PublisherBase(node_base, topic_name, type_support, rcl_publisher_get_default_options(), callbacks_, true)

ros2_humble/install/include/rclcpp/rclcpp/publisher_base.hpp:77:3: note:   candidate expects 4 arguments, 6 provided

`

facontidavide, 2024-02-26:
Reminder that you should use branch **main**.

Check if this PR (just merged) helps: #82

Tiqo01, 2024-02-26:
Thanks, it works !

salmagro, 2024-02-27:
> I just tested and it works just fine. I don't know how to help you

Everything works fine also for me on Debian 11 and 12, using humble. Thanks.

closed 2024-02-27 by facontidavide