PlotJuggler/plotjuggler-ros-plugins #83: ROS2 Topic Re-Publisher segfaults

doisyg, 2024-03-02:
Environment:
OS: Ubuntu 22.04
ROS2 Iron lastest binaries as of 02/02/2024
PlotJuggler and ros msg and pluggin lastest source as of 02/02/2024
Cyclone DDS

https://github.com/PlotJuggler/plotjuggler-ros-plugins/assets/15727892/df5575a6-4e56-4c20-b94d-e5381c73139a

```
$ ros2 run plotjuggler plotjuggler 
Loading compatible plugins from directory:  "/home/gd/ws_plotjuggler/install/plotjuggler_ros/lib/plotjuggler_ros"
"libDataLoadROS2.so is a DataLoader plugin"
"libDataStreamROS2.so is a DataStreamer plugin"
[WARN] [1709379774.941604065] [rclcpp]: logging was initialized more than once
"libTopicPublisherROS2.so is a StatePublisher plugin"
Number of plugins loaded:  3 

Loading compatible plugins from directory:  "/home/gd/ws_plotjuggler/install/plotjuggler/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamMQTT_Mosquitto.so is a DataStreamer plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserDataTamer.so is a MessageParser plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
QtAV 1.13.0(Mar 17 2022, 15:37:53)
Multimedia framework base on Qt and FFmpeg.
Distributed under the terms of LGPLv2.1 or later.
Shanghai, China Copyright (C) 2012-2019 Wang Bin (aka. Lucas Wang) wbsecg1@gmail.com
Donate: http://qtav.org/donate.html
Source: https://github.com/wang-bin/QtAV
Home page: http://qtav.org
"libPublisherVideoViewer.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
"libToolboxRemote.so is a Toolbox plugin"
Number of plugins loaded:  17 

Loading compatible plugins from directory:  "/home/gd/.local/share/PlotJuggler"
Number of plugins loaded:  0 

The rosbag loaded the data in  2210  milliseconds
Stack trace (most recent call last):
#31   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f1722029d8f]
#30   Source "/home/gd/ws_plotjuggler/src/PlotJuggler/plotjuggler_app/main.cpp", line 486, in main [0x555f83195110]
        483:   manager_message.get(request_message);
        484: 
        485: 
      > 486:   return app.exec();
        487: }
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f1722ac0cf3, in QCoreApplication::exec()
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f1722ab875a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f1722b130b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7f172191e3e2, in g_main_context_iteration
#25   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7f1721976257, in 
#24   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7f1721920d3a, in g_main_context_dispatch
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3", at 0x7f171c4bfd6d, in 
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7f1722f16a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7f1722f41306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f1722ab9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f172376c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[... 63 lines cut ...]
    | Source "/usr/include/c++/11/bits/alloc_traits.h", line 516, in construct<GenericPublisher, rclcpp::node_interfaces::NodeBaseInterface*, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const rosidl_message_type_support_t&>
    |   514: 	{
    |   515: #if __cplusplus <= 201703L
    | > 516: 	  __a.construct(__p, std::forward<_Args>(__args)...);
    |   517: #else
    |   518: 	  std::construct_at(__p, std::forward<_Args>(__args)...);
    | Source "/usr/include/c++/11/ext/new_allocator.h", line 162, in GenericPublisher
    |   160: 	construct(_Up* __p, _Args&&... __args)
    |   161: 	noexcept(std::is_nothrow_constructible<_Up, _Args...>::value)
    | > 162: 	{ ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    |   163: 
    |   164:       template<typename _Up>
      Source "/home/gd/ws_plotjuggler/src/plotjuggler-ros-plugins/src/TopicPublisherROS2/generic_publisher.h", line 32, in create [0x7f16f4328d71]
         29: #ifdef ROS_HUMBLE
         30:   : rclcpp::PublisherBase(node_base, topic_name, type_support, rcl_publisher_get_default_options())
         31: #else
      >  32:   : rclcpp::PublisherBase(node_base, topic_name, type_support, rcl_publisher_get_default_options(), callbacks_, true)
         33: #endif
         34:   {}
#1    Object "/opt/ros/iron/lib/librclcpp.so", at 0x7f172418dad6, in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_support_t const&, rcl_publisher_options_s const&, rclcpp::PublisherEventCallbacks const&, bool)
#0    Object "/opt/ros/iron/lib/librclcpp.so", at 0x7f17240f0273, in 
Segmentation fault (Signal sent by the kernel [(nil)])
[ros2run]: Segmentation fault

```

fbwogn, 2024-05-27:
Same problem occurs. Did you solve it?

nachovizzo, 2024-06-30:
I'm also facing this issue now :eyes:

nachovizzo, 2024-06-30:
So I couldn't find the root cause, but it's unlikely a plotjuggler problem.

What I did to hack this in the meantime to use this functionality is to comment out this line in `rclcpp`
https://github.com/ros2/rclcpp/blob/8de4b90512469d74802871e6c88db2e3135720e7/rclcpp/src/rclcpp/publisher_base.cpp#L55

```cpp
event_callbacks_(event_callbacks)
```

This line invokes a copy construction of each callback object defined within the `PublisherEventCallbacks event_callbacks_;` member, this is defined as follow (in https://github.com/ros2/rclcpp/blob/8de4b90512469d74802871e6c88db2e3135720e7/rclcpp/include/rclcpp/event_handler.hpp#L15):
```cpp
using QOSDeadlineOfferedInfo = rmw_offered_deadline_missed_status_t;
using QOSLivelinessLostInfo = rmw_liveliness_lost_status_t;
using QOSOfferedIncompatibleQoSInfo = rmw_offered_qos_incompatible_event_status_t;
using IncompatibleTypeInfo = rmw_incompatible_type_status_t;
using MatchedInfo = rmw_matched_status_t;

using QOSDeadlineOfferedCallbackType = std::function<void (QOSDeadlineOfferedInfo &)>;
using QOSLivelinessLostCallbackType = std::function<void (QOSLivelinessLostInfo &)>;
using QOSOfferedIncompatibleQoSCallbackType = std::function<void (QOSOfferedIncompatibleQoSInfo &)>;
using IncompatibleTypeCallbackType = std::function<void (IncompatibleTypeInfo &)>;
using PublisherMatchedCallbackType = std::function<void (MatchedInfo &)>;
/// Contains callbacks for various types of events a Publisher can receive from the middleware.
struct PublisherEventCallbacks
{
  QOSDeadlineOfferedCallbackType deadline_callback;
  QOSLivelinessLostCallbackType liveliness_callback;
  QOSOfferedIncompatibleQoSCallbackType incompatible_qos_callback;
  IncompatibleTypeCallbackType incompatible_type_callback;
  PublisherMatchedCallbackType matched_callback;
};
```

When trying to copy the `GenericPublisher::callbacks_` into the rclcpp base publisher, the segfault does happen in the copy constructor of the `std::function`  objects, so there is something fishy there

https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/d0692ad5fb7ea9f314f40816ea9e0a4fe55f8397/src/TopicPublisherROS2/generic_publisher.h#L61

As reference, the implementation of the gcc11 copy constructor of `std::function`:
```cpp
/**
 *  @brief %Function copy constructor.
 *  @param __x A %function object with identical call signature.
 *  @post `bool(*this) == bool(__x)`
 *
 *  The newly-created %function contains a copy of the target of
 *  `__x` (if it has one).
 */
function(const function& __x) : _Function_base()
{
  if (static_cast<bool>(__x))
  {
    __x._M_manager(_M_functor, __x._M_functor, __clone_functor); // <---- crashes here!!!
    _M_invoker = __x._M_invoker;
    _M_manager = __x._M_manager;
  }
}
```

ShkarupaDC, 2026-01-06:
Hi! I have also encountered this issue and came up with one more ugly solution:
1. Do not add the `callbacks_` field as an attribute of the `GenericPublisher`
2. Create the `PublisherBase` from scratch when initializing the `PublisherBase` by passing `{}` instead of the `callbacks_`

@facontidavide, could you please explain why you added the `callbacks_` field to the `GenericPublisher`? And how ugly is my solution?