PlotJuggler/plotjuggler-ros-plugins #87: ros-rolling-plotjuggler-ros not available for latest rolling

bijoua29, 2024-05-09:
ros-rolling-plotjuggler-ros is not available for latest rolling sync on 2024-04-30. The debian package doesn't exist.

bijoua29, 2024-05-14:
I tried compiling from the ros plugins from source and get the following errors:

Starting >>> plotjuggler_ros
--- stderr: plotjuggler_ros                                     
.../src/plotjuggler-ros-plugins/src/DataStreamROS2/datastream_ros2.cpp:10:10: fatal error: rosbag2_transport/qos.hpp: No such file or directory
   10 | #include "rosbag2_transport/qos.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/DataStreamROS2.dir/build.make:97: src/CMakeFiles/DataStreamROS2.dir/DataStreamROS2/datastream_ros2.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:223: src/CMakeFiles/DataStreamROS2.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
.../src/plotjuggler-ros-plugins/src/DataLoadROS2/dataload_ros2.cpp: In member function ‘virtual bool DataLoadROS2::readDataFromFile(PJ::FileLoadInfo*, PJ::PlotDataMapRef&)’:
.../src/plotjuggler-ros-plugins/src/DataLoadROS2/dataload_ros2.cpp:196:53: error: ‘using std::__shared_ptr_access<rosbag2_storage::SerializedBagMessage, __gnu_cxx::_S_atomic, false, false>::element_type = struct rosbag2_storage::SerializedBagMessage’ {aka ‘struct rosbag2_storage::SerializedBagMessage’} has no member named ‘time_stamp’
  196 |     const double msg_timestamp = 1e-9 * double(msg->time_stamp);  // nanoseconds to seconds
      |                                                     ^~~~~~~~~~
gmake[2]: *** [src/CMakeFiles/DataLoadROS2.dir/build.make:97: src/CMakeFiles/DataLoadROS2.dir/DataLoadROS2/dataload_ros2.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:195: src/CMakeFiles/DataLoadROS2.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< plotjuggler_ros [6.31s, exited with code 2]

christophfroehlich, 2024-07-31:
This is still not released for noble
https://build.ros2.org/job/Rbin_uN64__plotjuggler_ros__ubuntu_noble_amd64__binary/124/console fails with 

> 08:36:24 /tmp/binarydeb/ros-rolling-plotjuggler-ros-2.1.1/src/ros_parsers/ros2_parser.cpp:11:10: fatal error: PlotJuggler/fmt/core.h: No such file or directory