PlotJuggler/plotjuggler-ros-plugins #91: humble branch compilation failed

Heaven-Master, 2024-09-13:
..../ws_plotjuggler/src/plotjuggler-ros-plugins/src/ros_parsers/ros2_parser.cpp:
11:10: fatal error: PlotJuggler/fmt/core.h: file or directory not found
   11 | #include <PlotJuggler/fmt/core.h>

I fixed it by changing this line to

`#include <PlotJuggler/contrib/fmt/core.h>`

salmagro, 2024-12-10:
It seems that the following commit https://github.com/PlotJuggler/plotjuggler-ros-plugins/commit/f2c14b147182654c35cb711f78856385bd831e60 needs to be considered also on the `humble` branch.

PS. Please try to be more descriptive when creating a new issue. Provide necessary information to reproduce the problem and possible solutions.

christophfroehlich, 2025-05-19:
Oh sorry, I haven't looked further after I found this issue. Thanks for fixing!
You know for sure better than I do: But you tagged the repo but haven't released it to [rosdistro](https://github.com/ros/rosdistro/blob/5f3d15702668ae0a4e83b64bd3f45c3e6f023580/humble/distribution.yaml#L6846) yet, right?

facontidavide, 2025-05-30:
Submitting a PR